Macrobase-tech/MKS-DLC32 #1: Running MKS DLC32 V2.0 without touch screen

deltazerorsan, 2021-12-04:
Hi there,

I'm configuring my new board MKS DLC32 V2.0 for use with a CNC machine.

I just load de CNC version of the firmware using MKSLaserTool; everything correct.

But then I realized that there is no WiFi menu in this version of MKSLaserTool. My intention is to use the board directly attached via USB to the workshop computer and supervise it via WiFi through a browser or via your mobile app, so no touchscreen at all.

1. How can I configure the WiFi from the UGS (universal gcode sender) console?
2. where can I find your mobile App?

Thanks in advance!

Rom

Macrobase-tech, 2021-12-06:
Hello Rom,
The wifi function is available with MKS TS24/TS35 display.
If without display,just DLC32,not work so far.
1.unable to enable by the UGS.
2.Please search MKSLASER on Google store,and download.

Best Regard!

主题: [Macrobase-tech/MKS-DLC32] Running MKS DLC32 V2.0 without touch screen (Issue #1)

deltazerorsan, 2021-12-06:
What a shame, the new board is really nice and, with the addition of the ESP32, very powerfull.

Why MKSLaserTool has not the WiFi menu? It is present in the documentation.

Is there the possibility to activate and configure WiFi via SD card?

Please let me know about any solution you can find.

Thanks!

Best regards,

Rom

RandomHacks-Git, 2021-12-06:
Hi, check this https://github.com/makerbase-mks/MKS-DLC32/issues/12

deltazerorsan, 2021-12-06:
> Hi, check this [makerbase-mks/MKS-DLC32#12](https://github.com/makerbase-mks/MKS-DLC32/issues/12)

Thanks a lot! The pointer is really valuable!

Rom

closed 2021-12-06 by deltazerorsan

Macrobase-tech, 2021-12-08:
Thank you for the Github link sharing,Makerbase is our supplier.
Activate and configure WiFi via SD card will be available in the feature.