Macrobase-tech/MKS-DLC32 #4: Air assist (M8,M9)

ruslanbanah, 2022-02-08:
Hi folks.  I assembled a laser cnc based on this motherboard.  everything works great!  But I needed air assistance… could you help with that?  which pin is controlled by GRBL commands M8 and M9

Macrobase-tech, 2022-02-09:
here is M8 control on cnc.

Best Regard!

主题: [Macrobase-tech/MKS-DLC32] Air assist (M8,M9) (Issue #4)

ruslanbanah, 2022-02-09:
@Macrobase-tech 
Sorry, but I don't see any attachments. Pls send me to email: ruslanbanah@gmail.com

guilobo, 2022-09-07:
can you send to me as too? gui.wolflobo@gmail.com

Macrobase-tech, 2022-10-11:
hello,what file I should send?

 
can you send to me as too? ***@***.***

bobbyalata, 2022-12-31:
could you please explain how the release of M7 and M8 is done? where should i connect the air assistant to
MKS-DLC32

AndreyChervyakov, 2023-02-08:
Air Assist. In manual I find "SCL pin can be used as a cooler function", it's I2C interface on plate. It's a 3.3V signal i think, but maybe 5V. In LigthBurn software you can choose M7 or M8 in device setting. From Youtube one man from channel plottmania.de show some thinks. Connection via MOSFET to I2C of board. I hope it work so.

Macrobase-tech, 2023-02-09:
Right.
3.3v pin is to controls the air assist. Used as a control signal, not for load.

1005918473
***@***.***