Macrobase-tech/MKS-DLC32 #5: Suggestions for improvement.

IvanPotapow, 2022-02-24:
Hello developers!
DLC32 CNC V2.0 firmware (20220105) needs to be finalized.
1) in the "Control" section, add the command "Search for the starting position" (Hard limits).
2) in the "Control" section, add the "Z probe" command.
3) add beep.
4) downloading to an sd card with a Latin file name G-code does not work. (before the repair, write it on the project page, as a warning to users).
5) I want to see the display of the model on the screen before processing the part (as on a 3d printer). thank you!!!

Macrobase-tech, 2022-02-25:
Thank you for the kindness suggests.
Point 1.2.4.CNC endstops, Z probe as well as SD card will add in the feature.
3.Beep had already added.
5.It 's based on software which most are closed soure,so hard to develop so far.

IvanPotapow, 2022-02-25:
OK, we are waiting for the firmware update.

Macrobase-tech, 2022-02-25:
Thank you for follow up.