Macrobase-tech/MKS-DLC32 #6: Problem with Pwm Spindle

phildel71, 2022-04-03:
Hello,
I'm facing a problem with the pwm output on the 2 pin Spindle. 
Configuration
Firmware: V2.0.8_H35_20220105_N_CNC
Vin: 24V
Grbl parameter $30 Max spindle: 10000
If I put the command M03S1000, I measure an output voltage of 24V, I have to set $30 to 100000 to obtain a range 0-24V for a range of spindle speed of 0-10000
Unfortunately my spindle has a max speed of 12000 and it is not possible to set up than 100000 for the parameter $30 of Grbl

Macrobase-tech, 2022-04-06:
$30 1000,
It‘s output ratio, not specific speed.
If set 1000 and send $30 1000, it means 100% output. Not the speed. 100% output may be output according to max speed.

When send $30 1000,output max 24v.

Best Regard!

主题: [Macrobase-tech/MKS-DLC32] Problem with Pwm Spindle (Issue #6)

phildel71, 2022-04-06:
Hello,
Sorry but this behaviour is not what is described in the documentation of Grbl for the parameter $30. I already have two others boards which uses it and the value of the PWM is a value based on the value of max spindle speed which is the real max RPM of the spindle.
For example if I have $30=12000 and the programmed spindle speed is S6000, the PWM output is 12V for a  VIN of 24V (or 2.5V for some boards with a max output of 5V).

Regards

Macrobase-tech, 2022-04-07:
It's available that way.
when you set 1000 and send command,

 it will change to 1200 and output.