Macrobase-tech/MKS-DLC32 #8: The screen MKS TS35-R V2.0 does not respond to commands

zmei46, 2022-04-27:
Hello. My screen doesn't respond well to touch. Firmware version V2.0.8_H35_20220105_N_CNC. When you press the buttons, it does not respond to them. Sometimes when pressed, other buttons work. The button to return to the previous menu does not react at all. Tried different firmware, nothing changes. Help what to do?
Здравствуйте. У меня экран плохо реагирует на нажатия. Версия прошивки V2.0.8_H35_20220105_N_CNC. Когда нажимаешь кнопки он не реагирует на них. Иногда при нажатии срабатывают другие кнопки. На кнопку вернуться в предыдущее меню вообще не реагирует. Пробовал разные прошивки, ничего не меняется. Помогите, что делать?

Macrobase-tech, 2022-04-28:
​Hello,please try below.
1.steady the yellow wiring of TS35 back.
2.loose the frame of TS35.

主题: [Macrobase-tech/MKS-DLC32] The screen MKS TS35-R V2.0 does not respond to commands (Issue #8)

zmei46, 2022-04-28:
Hello. Your advice helped - loosened the frame and everything works as it should without any complaints! Thanks for the advice!
Здравствуйте. Ваш совет помог - ослабил рамку и все работает как надо без нареканий! Спасибо за совет!

Macrobase-tech, 2022-04-29:
That's great to hear that.