Macrobase-tech/MKS-DLC32 #9: Problema pwm min

distebia, 2022-07-03:
salve, ho un problema con una mks dlc32 v2.1 e firmware 2.08 e 2.10 per laser normale board 2.0. il problema è che non riesco ad accendere il laser sotto il 10%. con arduino montato nella stessa macchina andavo a variare il parametro spindle_pwm_min_value contenuto in config.h e impostavo il valore di 25 per aggiungere mV all'uscita pwm, per avere la minima accensione.
i miei parametri su arduino uno sono $30=255, $31=0. Con arduino configurato in questo modo accendo il laser al 1%. Con mks dlc32 v.2.1 non ho modo di fare la stessa cosa in quanto non ho il config.h. Ho trovato il parametro $28 che, momentaneamente, ho lasciato a 1000, $35 che dovrebbe essere l'equivalente di spindle_pwm_min_value ma variandolo non ottengo differenze. Ho anche impostato $31=100 ma accendo sempre il laser intorno al 10% e, anche variando $30, $31, $35, $28 non ottengo variazioni di emissione pwm. 
tutto questo sia con lasergrbl che con lightburn (LB configurato come da $30). è come se la scheda memorizza i valori dati ma non li esegue, o li esegue entro un certo limite.

distebia, 2022-07-07:
esiste un firmware per laser normale che mi da la possibilità di variare i parametri inerenti l'uscita pwm? grazie

Macrobase-tech, 2022-07-09:
Sorry,our DLC32 firmware doesn't have.
Official GRBL can be a try.

distebia, 2022-07-09:
un conto è un firmware chiuso dove non posso cambiare indirizzi hardware o velocità processore ecc... un conto è impedire dei settaggi basilari. da come ho capito tutti dobbiamo avere lo stesso alimentatore che si accende allo stesso modo, accetta la stessa frequenza di pwm. Se ho comprato il vostro prodotto completo di monitor lcd perchè devo mettere un altro firmware perdendo altri benefici?

closed 2023-09-09 by distebia