Macrobase-tech/MKS-DLC32 #10: The X and Y axes are rotated by 45 degrees

SergChel, 2022-07-16:
With "MKS DLC32+TS24" firmware  for the laser, the X and Y axes are rotated by 45 degrees. Instead of moving in a straight line, the laser goes diagonally. Instead of moving diagonally, the laser goes straight. If I download the firmware for the CNC, the bug disappears. I have problem with the V2.08_H24_20211223_C.bin and V2.08_H24_20220107_C.bin  firmware for from this site.

Macrobase-tech, 2022-07-18:
Hello,please try below firmware.
https://github.com/Macrobase-tech/MKS-DLC32/tree/main/MKS%20DLC32%2BTS24/Firmware/CNC

主题: [Macrobase-tech/MKS-DLC32]  The X and Y axes are rotated by 45 degrees (Issue #10)

SergChel, 2022-07-18:
> Hello,please try below firmware. https://github.com/Macrobase-tech/MKS-DLC32/tree/main/MKS%20DLC32%2BTS24/Firmware/CNC
> […](#)

When I use this firmware laser does not works in the dinamic mode, it works only in the constant power mode.

Macrobase-tech, 2022-07-20:
My colleague had sent another fw to fit your machine.

SergChel, 2022-07-20:
Thank you. Could you give me link on this firmware please?

sernik68n, 2025-03-12:
У меня такая же проблема с MKS DLC32 v.2.1 Подскажите как исправить. Буду очень благодарен.