Macrobase-tech/MKS-DLC32 #18: Code for CNC32

marankos, 2023-01-13:
For CNC32, I cant find for it.  The DLC32 is compatible to CNC32?

Macrobase-tech, 2023-01-13:
That's right,DLC32 firmware is also for CNC32.

主题: [Macrobase-tech/MKS-DLC32] Code for CNC32 (Issue #18)

marankos, 2023-01-13:
One question, for z probing assigned pin is different one is IO4 , and other is IO24. The firmware knows itself the different IO pin and operate witj different IO pin????

Regards

主题: [Macrobase-tech/MKS-DLC32] Code for CNC32 (Issue #18)

Macrobase-tech, 2023-01-13:
We recommand to use port pin on our CNC control board.

1005918473
***@***.***

 
 
 
 
 
 ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------
 主题:&amp;nbsp;[Macrobase-tech/MKS-DLC32] Code for CNC32 (Issue #18)

marankos, 2023-01-13:
Dear 

My expression of the previous question is not clear. I apologize to you.
I saw one picture that the zero probing IO pin is described 'IO4'
On other picture the zero probing IO pin is assigned 'IO22' with SA? pin or SB pin
But most of the 8bit control card with GRBL1.1 is using 'A5'

So I was asked to you that CNC32 hardware use the dual address for zero probing or which is the correct IO pin.

Sorry again to make the confusion again.

Regards

Quilan

Macrobase-tech, 2023-01-14:
probe is IO22
CNC32 and DLC32 are same

1005918473
***@***.***

 
 
 
 
 
 
 
 
 
 
 
 
 
 
 &amp;nbsp;
 
 
 
 
 主题:&amp;nbsp;Re: [Macrobase-tech/MKS-DLC32] Code for CNC32 (Issue #18)
 
 
 
 
 
 
 
 
 
 
 ------------------&amp;amp;nbsp;原始邮件&amp;amp;nbsp;------------------
 主题:&amp;amp;nbsp;[Macrobase-tech/MKS-DLC32] Code for CNC32 (Issue #18)

AndreyChervyakov, 2023-02-08:
What is the difference between "normal" and "core XY" software .bin files from laser folder?

Macrobase-tech, 2023-02-09:
Hello,different machine structure.

1005918473
***@***.***