Macrobase-tech/MKS-DLC32 #19: Mks dlc32 ts2. 4 cnc

Loloxy, 2023-03-12:
Hello! After installing the command
$1=255 coordinates are not reset. Can this be fixed? And will a homing button be added?

Macrobase-tech, 2023-03-15:
There is homing button on display.
It doesn't support self-customize button so far

Subject: [Macrobase-tech/MKS-DLC32] Mks dlc32 ts2. 4 cnc (Issue #19)

Loloxy, 2023-03-15:
What about the $1 team? If you set the value to 255, the possibility of zeroing the coordinates disappears

Macrobase-tech, 2023-03-15:
Set $1=5  milliseconds  Step idle delay

Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay

1005918473
***@***.***

Loloxy, 2023-03-15:
That's how I understand it. Why does the possibility of zeroing the coordinates disappear? My machine design requires the Motors to be on hold at all times.

Macrobase-tech, 2023-03-17:
At the beginning of powering on, if motor no moving,default motor unlock . As long as the motor has moved, it will lock up."

1005918473
***@***.***

Loloxy, 2023-03-17:
You probably do not understand what I want from you. The Z axis is heavy and when the motors are off, $1 =5, it falls from its own weight, with the command $1 =255, the ability to reset the coordinates disappears. Is it possible to fix it in the firmware or not?

Macrobase-tech, 2023-03-17:
Set a value between 5-255.

 And the save,it will be fixed 

1005918473
***@***.***