Macrobase-tech/MKS-DLC32 #21: MKS DLC32 V2.1 board, the PWM output signal decreases when both axes are working

SoliJhon, 2023-03-20:
Hi all!
On my MKS DLC32 V2.1 board, the PWM output signal decreases when both axes are working, the higher the speed - more than 10 mm / s, the more the voltage drops between the S and TTL pins.
Diver TMC2209. Nema 17HS4801.
Everything is fine when it cuts while moving along one axis - vertically or horizontally. If you lower the speed to 10 mm / s, then the voltage drop is almost not noticeable. There are no problems with Gcod, since it works correctly in another GRBL / controller. There is no interference or interference either, since I tried to disconnect everything - nothing has changed ... I tried changing the power supply from 12 to 24 volts, it did not help either. I thought something burned out in the board and bought a new one, with the new one the same problem ...
Nobody faced such a problem?
GRBL setting.
$0=10
$1=255
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=300.000
$25=1000.000
$26=250.000
$27=5.000
$28=1000
$30=1000.000
$31=0
$32=1
$38=1
$40=1
$100=80.000
$101=80.000
$102=80.000
$103=100.000
$104=100.000
$105=100.000
$110=3000.000
$111=3000.000
$112=6000.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=500.000
$121=500.000
$122=500.000
$123=200.000
$124=200.000
$125=200.000
$130=450.000
$131=450.000
$132=50.000
$133=300.000
$134=300.000
$135=300.000

Macrobase-tech, 2023-03-22:
Did you modify some speed parameters  on firmware?

Please try to update default firmware without modifying any parameters.

Belt or screw motor? If it is a screw motor, ensure the smoothness of the screw.

Subject: [Macrobase-tech/MKS-DLC32] MKS DLC32 V2.1 board, the PWM output signal decreases when both axes are working (Issue #21)

SoliJhon, 2023-03-22:
1. No.
2. The problem remains even with the new firmware without changing the parameters.
3. Belt.

Macrobase-tech, 2023-03-25:
After tested,work well.
good to try other drivers

1005918473
***@***.***

  
No.
 

 
Belt.