Macrobase-tech/MKS-DLC32 #22: Сборка проэкта в bin

Babaika1987, 2023-04-05:
Здравствуйте можите написать мануал по сборке проэкта в bin файл. Я пробывал собрать но она не работает.

Macrobase-tech, 2023-04-08:
Hello,we release source code for self-develop
https://github.com/Macrobase-tech/MKS-DLC32/blob/main/DLC32-Firmware%20Code.zip

Subject: [Macrobase-tech/MKS-DLC32] Сборка проэкта в bin (Issue #22)

Babaika1987, 2023-04-08:
Я позже пришлю вам видео что-бы вы могли видеть сами что wi fi дома
работает. На дрогой прошивки он работает, а на этой нет

Macrobase-tech, 2023-04-08:
A video is goog for check

1005918473
***@***.***

 
 
 &gt; Hello,we release source code for self-develop 
 &gt; ------------------&amp;nbsp;Original&amp;nbsp;------------------ 
 &gt; From: "Macrobase-tech/MKS-DLC32" ***@***.***&amp;gt;; 
 &gt; Date:&amp;nbsp;Wed, Apr 5, 2023 11:38 AM 
 &gt; ***@***.***&amp;gt;; 
 &gt; ***@***.***&amp;gt;; 
 &gt; Subject:&amp;nbsp;[Macrobase-tech/MKS-DLC32] Сборка проэкта в bin (Issue #22) 
 &gt; Здравствуйте можите написать мануал по сборке проэкта в bin файл. Я 
 &gt; пробывал собрать но она не работает. 
 &gt; ID: ***@***.***&amp;gt; 
 &gt; ***@***.***&gt;

closed 2023-04-09 by Babaika1987