Macrobase-tech/MKS-DLC32 #24: Zero Point Search

Babaika1987, 2023-04-07:
https://github.com/Macrobase-tech/MKS-DLC32/blob/main/MKS%20DLC32%2BTS35/Firmware/Laser/Normal/Board_V2.0/V2.0.8_H35_20220105_N.bin

When the machine is looking for a zero point, it makes a 1mm indent from the endpoints and should reset these coordinates, but it does not do this and in the coordinates the state is 1mm in X and Y if you press home, it returns to zero and presses the end point.

Macrobase-tech, 2023-04-08:
Sorry,i can't understand the issue you desrcibe.
Is it possible to show in video?

Subject: [Macrobase-tech/MKS-DLC32] Zero Point Search (Issue #24)

 
https://github.com/Macrobase-tech/MKS-DLC32/blob/main/MKS%20DLC32%2BTS35/Firmware/Laser/Normal/Board_V2.0/V2.0.8_H35_20220105_N.bin

Babaika1987, 2023-04-08:
Ok

Babaika1987, 2023-04-09:
Я решил проблему прошивкой 2.30ver

closed 2023-04-09 by Babaika1987

Macrobase-tech, 2023-04-10:
That's wonderful

1005918473
***@***.***