Macrobase-tech/MKS-DLC32 #26: Comment choisir le bon fichier entre: Courroie ou Vis sans fin ?

Couzen, 2023-04-14:
Bonjour a tous.
Je suis nouveau, et je suis aussi un amateur de couleur bleu.
J'ai acheté il y a quelques temps un kit de laser pour graver.
- Model: Wainlux JL3.
Je voudrais faire une mise a niveau pour le faire évoluer un peux, et que je puise en faire de même, moi aussi.
J'ai donc acheté quelques accessoires pour le faire:
- Un Laser: Sculptfun S9 - (Type: S9DNO10A).
- Une carte mère: Makerbase MKS DLC32 V2  --  32bits ESP32 Carte WIFI TS35  + (3x Drivers: A4988  -- [HR4988so]).
- 2 interrupteurs de fin de course mécanique normalement ouvert, avec lumière de déclenchement. Et fiche de connexion a 3 contactes (V, G, S) - (3,3V - 5V).
- Le châssis du JL3 et fournis avec 2 moteurs pas a pas, sans aucune indications. (Dimensions des 2 moteurs: 42 x 42 x 32 mm).
- Le déplacement de la tête, ce fait avec une courroie. Et non par une vis sans fin.
- Les poulies ont 20 dents et un diamètre 16 mm.
- J'aimerais utiliser Laser GRBL avec tout ceci.
Voila, je crois que j'ai fait le tour des informations importante.

A présent je vais pouvoir vous dire ce qui me bloque.
Je cherche a télécharger le bon fichier. Et je voudrais savoir le quelle choisir ? 
Y a t il une différence entre un système avec courroie et celui avec vis sans fin ? 
Si oui, le quelle choisir SVP ? Merci. 

# Exemple: 
-> MKS DLC32+TS35.
-> Micrologiciel.
-> Laser.
-> Normal  (ou)  CoreXY  ????
-> Suivant  ???? :     Je dois prendre le quelle ? Quelle suite prendre....
Je suis un peut perdu ... 

Et dans la perspective qu'une suite, j'aurais besoin d'une personne pour m'aider a faire le paramétrage de ma machine avec les accessoires que j'ai listé plus haut.
Bien sûr, si vous avez besoin d'autre informations, ou que je face des testes, je pourrais les faire ou vous les donner sans problème.
Ah oui, comme rien n'est simple, je suis Français, et j'utilise un traducteur. 

Voila, je cherche donc une âme charitable qui pourrais me faire profiter de son savoir...
Merci a vous ...

Couzen, 2023-04-14:
How to choose the right file between:  Belt and Endless screw ?

Hello everyone.
I'm new and I'm also an amateur, color blue.
I bought some time ago a laser engraving kit.
- Model: Wainlux JL3.
I would like to make an "Upgrade - leveling" to make it evolve a little, and that I can do the same, too.
So I bought some accessories to do it:
- One Laser: Sculptfun S9 
- (Type: S9DNO10A).
- A motherboard: Makerbase MKS DLC32 V2 -- 32bits ESP32 WIFI Card TS35 + (3x Drivers: A4988 -- [HR4988so]).
- 2 normally open mechanical limit switches, with trip light. And connection plug with 3 contacts (V, G, S) - (3.3V - 5V).
- The JL3 chassis is supplied with 2 stepper motors, without any indications. (Dimensions of the 2 motors: 42 x 42 x 32 mm).
- The displacement of the head, this done with a belt.  And not by an endless screw.
- The pulleys have 20 teeth and a diameter of 16 mm.
- I would like to use Laser GRBL with all this.
 Well, I think I've covered the important information.

Now I can tell you what is blocking me. 
I am trying to download the correct file. 
And I would like to know which one to choose?
Is there a difference between a belt system and a worm gear system ?
If yes, which one to choose please ? THANKS.

# Example:
-> MKS DLC32+TS35.
-> Firmware.
->Laser.
-> Normal (or) CoreXY ????
-> Next ???? : Which one should I take? 
What next to take....
I'm a bit lost...

And in the perspective of a sequel, I would need someone to help me configure my machine with the accessories I listed above.
Of course, if you need other information, or that I face tests, I could do them or give them to you without problem.
Ah yes, as nothing is simple, I am French, and I use a translator.
Here, I am therefore looking for a charitable soul who could make me benefit from his knowledge...
Thank you ...

Macrobase-tech, 2023-04-14:
CoreXY firmware

 

 
  
 
 
 
 
 
 
 
 
  
 
 
-&gt; MKS DLC32+TS35.
 -&gt; Firmware.
 -&gt;Laser.
 -&gt; Normal (or) CoreXY ????
 -&gt; Next ???? : Which one should I take?

Couzen, 2023-04-14:
Bonjour,
Merci pour votre réponse.
Donc quand on a un châssis avec une transmission par courroie, il faut choisir : "CoreXY".
Vous pouvez me dire pour la suite ?

- MKS DLC32+TS35
- Micrologiciel
- Laser
- CoreXY
- Board_v1.0                                      (ou)                   -  Board_v2.0
- V2.0.8_H35_20211222_C.bin                                -  V2.0.8_H35_20220105_C.bin
- V2.0.8_H35_20220105_C.bin                                -  V2.10_H35_2022_0704_C_ZX_001.bin

Si j'ai bien compris c'est le C du "C.bin" qui indique l'option courroie ?!
Par contre je ne suis pas sur pour la suite.
La quelle choisir entre les deux option suivant Board 1 ou 2 ?
Et ensuite, le quelle choisir  dans les deux choix proposé... ?
Merci.

CoreXY firmware

Couzen, 2023-04-14:
Bonjour, 
Merci pour votre réponse.
Donc quand on a un châssis avec une transmission par courroie, il faut choisir : "CoreXY".
Vous pouvez me dire pour la suite ? 

- MKS DLC32+TS35
- Micrologiciel
- Laser
- CoreXY
- Board_v1.0.........................................................(ou)...................................................-  Board_v2.0
- V2.0.8_H35_20211222_C.bin..................................................................................-  V2.0.8_H35_20220105_C.bin
- V2.0.8_H35_20220105_C.bin..................................................................................-  V2.10_H35_2022_0704_C_ZX_001.bin

Si j'ai bien compris c'est le C du "C.bin" qui indique l'option courroie ?! 
Par contre je ne suis pas sur pour la suite. 
La quelle choisir entre les deux option suivant Board 1 ou 2 ? 
Et ensuite, le quelle choisir  dans les deux choix proposé... ? 
Merci.