Macrobase-tech/MKS-DLC32 #31: M5 or M30 at end of file stops machine before job done

Wiki-the-Tinkerer, 2023-07-29:
GRLB1.1
MKS DLC32 v2
pc, W11- bCNC, laserGRBL 5.01, Candle, grblControl etc.

Sending Gcode with M5 or M30 at end of file stops cnc machine before the job done. Tested with several different senders. Seems like controller would stop execution of the other still running commands when either M5 or M30 is confirmed with OK = program stops executing, and sends "idle", while sender keeps saying something like "98% completed". 

When commenting out only the M5 and M30 the programs work without problem.

Macrobase-tech, 2023-07-31:
hello,how the value you set on travel resolution step/mm?

Subject: [Macrobase-tech/MKS-DLC32] M5 or M30 at end of file stops machine before job done (Issue #31)

Wiki-the-Tinkerer, 2023-09-03:
Hi, Travel resolution : X1602, Y1245, Z809.5. 
However, these were altered many times, while calibrating the machine, and changes do not affect the issue of execution stopping when the M30 is reached in the sending to the que. 

This is not the sameas the "normal" random connection stops that happends all the time (possibly due to emi/power fluctuations - this clearly happens when the code up to end is recieved by the board. But.. not every time. Have not been able to identify any specific trigger.

Macrobase-tech, 2023-09-06:
Hello,we don't really understand this issueyet.
could you explain more detail?

1005918473
***@***.***

met-japan, 2023-09-11:
First, thanks Wiki-The-Tinkerer for figuring this out, yes, this happens if there is an M5 (Spindle Stop) command issued, has been very annoying as it's fairly standard in the footer. Deleting it from the footer seems to solve the problem but other versions of GRBL have had no problem. The temporary fix is to not issue spindle commands but MKS should fix this.

Wiki-the-Tinkerer, 2023-09-11:
Yes, it's a pretty normal thing to turn the spindle off, when the job is done. Besides, it saves a nice amount of fingers if you are able to turn off the spindle if you e.g. need to change tool... Or, then you'd either need to have a separate program for each tool, or a hardwired switch. 

My uneducated guess is that this is a matter of buffered sending of the commands VS syncronized sending. And spindle commands for some reason being priorized.

met-japan, 2023-09-12:
Wiki-the-Tinkerer, I agree with your synopsis. Standard G-Code is buffered (as is should be for read ahead computation), however M commands are executed as they are read instead of being added to the queue.

met-japan, 2023-09-14:
Potential Fix:
Set grbl $10=3
Have only tested with M3 and M5 but it worked fine running the whole programs both long and short. Will test tool changes over the next few days.

met-japan, 2023-09-16:
Tool changes also working fine. using TTL signal to control external relay for motor. So far so good.

Macrobase-tech, 2023-09-16:
Excellent

1005918473
***@***.***

Macrobase-tech, 2023-09-16:
turn off spindle by commend.
also can do it on display

1005918473
***@***.***

met-japan, 2023-09-18:
Things have been working well for about a week. So far all the programs I've run (about 50) have all run to completion including spindle control (on/off) and proper tool changes through M06.
Using bCNC as a front end.
I'm finding that the MKS-DLC32 firmware uses a hybrid of certain grblhal commands that are poorly documented.
This morning I had the controller freeze while issuing some simple codes via the MDI interface of bCNC which is worrying.
I still do not have confidence in this board.
I've downloaded the firmware source and will see if anything stands out.
I feel I'm taking two steps forward and one step back.
In CNC "almost working" is not good enough!

Wiki-the-Tinkerer, 2023-09-18:
Met-japan - thats strange.. Why would the status report options affect the execution of e.g. M5? 

In senders where its possible to affect - sort of "separate" - the sending and execution process the easiest dirty fix may be to avoiding buffering of the stream. (Sure as heck wont be removing the spindle stops from proper code, haven't got enough fingers ;)
 
E.g. in LaserGrbl, in the Settings / first tab Protocol /  Streaming Mode has to be set to "Syncronous" instead of the "Buffered" which otherwise would make more sense. You lose the whole fast buffer benefit, but it at least works.

Also noticed that some senders have inbuilt "Program End" code removal by default (M2,M30) - apparently to avoid resetting G92 offsets. Not saying what I think about that, stripping planned code without notification.. but it can mean that this error escapes without being noticed in some senders.  

General note: the board has become much more stable, when NOTHING except octocoupled endstop signals, and the stepper outs (to separate drivers) are connected. 

It seems that a major issue is the old usb chip (purely deducting here) that takes disturbance from anything, making usb very close to un-usable on a ny even slighlty dirty power lines. It also works good from SD, you just need to find some ancient FAT32able small enough card. (managed to get a 256mb card formatted after a couple hours of frustrating program downloads and tests. NOT as easy as you'd expect in newer Windows. WIN - There is a free edition of "EaseUS Partition master Free edition" - that can throw away the space of the card, to leave only 4GB, so that windows accepts to format it to FAT 32 that the card needs. Uh and Duh.)

Wiki-the-Tinkerer, 2023-09-18:
>In CNC "almost working" is not good enough! 
Yep, with bigger spindle motors (and lasers too i guess), not good enough can be really bad.
>I've downloaded the firmware source and will see if anything stands out.
Looking forward to any findings :)

ps met-japan, have you got the S speeds to work? ([separate issue thread](https://github.com/makerbase-mks/MKS-DLC32/issues/410)). If you look at the ingredients, and happen to cross the S speed calculation, maybe you could have quick a look if you have time?.. Methinks there is some major math error in there, in the translation/division of speed X to whatever resolution makes 1-100% in the board...  Should be simple maths, so probably just a mistake where no-one had considered spindles with higher speed. (eg. my small one has 24000r as optimal)

met-japan, 2023-09-19:
I think you've hit on something regarding the USB connection. After a "freeze" occurs the syslog shows a disconnect error of the ch341-uart (I use Linux).
Going to chase this down first as it could be crux of the problem.
I'll get back to you on the other issues you've mentioned.

met-japan, 2023-09-19:
Just confirmed, 45 minutes into a run syslog shows error USB disconnect, will try a different computer but if this problem persists because of the on board USB UART chip I'm going to donate the board to the local skeet range.

Macrobase-tech, 2023-09-19:
May try different GRBL software with different gcode

1005918473
***@***.***

met-japan, 2023-09-19:
Maybe try shipping with a proper and fully tested version of Grbl.

met-japan, 2023-09-19:
The main problem is with the USB connection, switching to a different computer with a known good powered USB port has solved the "freeze" issue. Ran an overnight test program monitoring the USB line without issue.
I had initially been using an RPI4 but the USB port does not have enough current for the MKS on board chip,switched to a laptop that I now has well powered ports and the issue went away. My other choice would be to use a powered hub but that just starts to get crowded.
The $10=3 is recommended in a few forums, doesn't make a lot of sense to me either but it works. All grbl supported M commands now work as expected.
I'm currently not using the PWM controller just M3 and M5 for motor control through the signal line of the TTL port with an external relay.
I do have a few different heads including a laser so will need to look into the PWM controller soon.
Next step is a day or two of regression testing to make sure all changes this far are truly working as intended.
My level of confidence in this board is now high enough to continue tweaking it.
Thanks for noting your suspicion of the USB being a possible problem, helped greatly in debugging.
FWIT I'm now connecting to the USB port with the original cable plus a  high quality 2 meter extension to reach the laptop with no sign of signal leakage.
I'm also thinking of switching from a 12v power supply to 24. I use external drivers for the stepper so probably not needed.
I need 24v 6a for the laser which I don't think the board can handle so will need to come up with a cunning plan.

Wiki-the-Tinkerer, 2023-09-28:
>I need 24v 6a for the laser which I don't think the board can handle so will need to come up with a cunning plan.

That should not be any problem, providing there is a working "right scaled" output - i.e. ttl 0v-5v. Apparently this works if S max setting is 1000. 
Then you connect the ttl to your laser drivers ttl input (sharing ground with laser power). May need a simple logic level shifter, or a fast transistor/octocoupler fix in between, if your laser drivers ttl input would not be 0-5v. With this board it seems a good idea to not run anything through the board, and octocouple everything possible, to minimize disturbances.

Macrobase-tech, 2023-10-06:
6a is high.please plug additional power to laser/spindle

1005918473
***@***.***