Macrobase-tech/MKS-DLC32 #32: Keeping the system in idle mode freezes the controller

iamyskim57, 2023-08-07:
I have 4  sets to run 3 axis CNC routers. I experience freezing of the controller when the the controller is left in idles state. 
All of them show similar problem. Here is the description of the problem:

Once the controller is activated to start cutting with uploaded GCODE data, it finishes the job regardless of the length of the code. 
But when the controller is left in idle state, it goes into state where no control is possible (neither on LCD panel nor WebUI) typically within few seconds to few minutes . The only way to excape from this state is resetting the controller. 
Even during freezed state, uploading of NC data is possible.

Do you have any suggestion to solve the problem.

Macrobase-tech, 2023-08-09:
what kind of the error ir mark on GRBL software?

Subject: [Macrobase-tech/MKS-DLC32] Keeping the system in idle mode freezes the controller (Issue #32)

iamyskim57, 2023-09-04:
I am connected through web connection. No error message is shown. Continuous message showing the status of the controller is shown even when the controller is freezed. In this frozen state uploading of .NC data is possible, but no movement command through the button on the LCD screen is possible. Also in this case no gcode command is possible.