Macrobase-tech/MKS-DLC32 #36: The problem is for the CNC firmware

Sim-26356, 2023-11-11:
Hello. The problem is for the CNC firmware . When using the HARD HOME function, only the X and Y axes are homing, the Z axis does not work. When using the GrblControl program, the machine coordinates of the Z axis are not determined. This prevents the MKS DLC-32 board from being used as a CNC machine. The machine is used in a normal configuration (not CoreXY). We really need a CNC machine. Please help with the firmware. We used the firmware MCS DLC32+TS35

Macrobase-tech, 2023-11-13:
try CNC corexy firmware

Subject: [Macrobase-tech/MKS-DLC32] The problem is for the CNC firmware (Issue #36)

Sim-26356, 2023-11-13:
We checked all four firmware versions from the MKS DLC32+TS35 folder, all equally did not help. The limit switch of the Z axis is in good condition, the firmware reacts with a rigid restriction when pressed. Posted here https://github.com/Macrobase-tech/MKS-DLC32/blob/main/DLC32-Firmware%20Code .zip the firmware source code for the CoreXY machine, can you provide a version of the source code for the Normal machine?

Macrobase-tech, 2023-11-14:
I sent you private mail

1005918473
***@***.***

Daget84, 2023-11-24:
I have the same problem.
help me solve please

vorotnykvs, 2025-01-22:
Hello, share the firmware for the CNC milling machine