Macrobase-tech/MKS-DLC32 #39: Issues with CNC32 Board

dyeatman, 2024-07-28:
I am installing the CNC32 board (not the DLC32) and have some questions:
1. I can move the Z axis using Jog but it does not move when I perform a home (X/Y axis homes fine)
2. I have an  EStop button, where does it connect on the CNC32?   
3. How do I enable continuous Jog Mode for local control?
4. Is the DLC32 firmware the only one that I can use on the CNC32 board?  The firmware I have shows as  GRBL 1.1 in UGS but I believe the latest version is 1.1H. 

Thank you.

Macrobase-tech, 2024-07-29:
Hi,here are.
The z-axis homing of software seems to be individually effective.
In software,homing is usually X/Y axis. 

Try sending this command once each to enable Z axis homing.
$Homing/Cycle0=XY
$Homing/Cycle1=Z

2. Connect to Reset pin.
3. The movement is just Jog Mode,without continuous.
4. Installed original MKS firmware is recommended

Subject: [Macrobase-tech/MKS-DLC32] Issues with CNC32 Board (Issue #39)

dyeatman, 2024-07-31:
Got Z Axis homing working now...Thank you.
My Z probe sensing does not stop/work after it contacts the probe plate.  
Used a VOM to test both wires and determined they are good and the connections look good (right pins according to the wiring diagram).
Changing $6=0 causes red Probe message to appear  on the controller state screen.

Macrobase-tech, 2024-08-01:
Z axis just for homing not for Z problem.
Z probe is wiring to another interface

1005918473
***@***.***

dyeatman, 2024-08-01:
I have the xyz probe connected to the probe pin per the wiring diagram, I assume that is correct.
I know the two are separate but didn't know one would affect the other.
Are you saying that enabling z axis homing is disabling the Z axis probe function?

If so,  I am willing to disable the Z axis homing to get the probe to work. 
What is the command for this?  I hate to keep asking these questions can I find the commands you reference somewhere?

Macrobase-tech, 2024-08-01:
Z axis in firmware is enable Z probe

1005918473
***@***.***

dyeatman, 2024-08-01:
Please forgive me but I am not understanding what you are saying.  Do I have the wrong firmware? 
Z Axis homing is working well and it sounds like you are saying that enabling Homing enables Z probe but it is not working...I am obviously missing something.

dyeatman, 2024-08-01:
I was just reading another post where it was said the probe should be on IO22 which is the SA pin but the wiring diagram shows the probe being on the SB pin.  Should I be using SA instead of SB?

dyeatman, 2024-08-01:
It turns out the probe should be on SA **_not_** SB. The  wiring diagram on the Alibaba page (link below) is **_wrong_** where it clearly shows the SB pin as being the probe pin.  Can you get this fixed?  Thank you for your help.
![CNC32 Wiring Connections]
(https://github.com/user-attachments/assets/ae64fc54-9286-4041-9f93-3f97157eec88)

Macrobase-tech, 2024-08-02:
Sorry for the mistake wiring,you are right,
Z probe is connecting to SA,SB is for Air assistance.
We will update the wiring later,thank you.

1005918473
***@***.***